cnlohr/esp82xx #10: Troubleshoot reboot?

cnlohr, 2016-08-19:
I, on my wifi chipset, have a hard time reconnecting to ESPs after they've been usbburned or netburned a few times.  They need a hard power cycle.  I'm not sure why, though.  Is anyone else having this trouble?

brainstorm, 2016-08-24:
How many times did you try to re-burn per session? Have been doing `netburn` regularly (20+ times in a row now) after getting my docker setup right and my ESP-05 module is a happy camper so far.

Btw, your software is awesome, love the short iterations with `make netburn`, reeeally sweet!

cnlohr, 2016-08-24:
After only five iterations things start to seize up.  It could also be a side effect of the awful drivers on my Atheros qca9377.  I am || this close to swapping it out for an Intel.  If you aren't having trouble with it, might just be me.  If someone else has problems, please re-open this thread.

closed 2016-08-24 by cnlohr

brainstorm, 2016-08-24:
Actually just checked and I have this one lying around that I haven't used for years (nor plan to):

http://www.intel.com/products/wireless/prowireless_mobile.htm

Not `N` nor `AC` fancy new stuff, but might do the trick for you. Let me know if you are interested and I can ship it to you ;)

cnlohr, 2016-08-24:
I actually replaced the intel in my old laptop, which was that one, so I have that hanging around.  I think the problem was that this new laptop used a totally different connector.  Don't quite remember.