cnlohr/esp82xx #3: Need to add con-f-use to credits

cnlohr, 2016-08-19:
Not sure what you want to go by in the copyright or on the webpage.

but... @con-f-use is gonna have to get some credit here!

con-f-use, 2016-08-19:
Nah, I'm good. The janitor of CERN doesn't get to be on the papers ;-) You guys wrote all the heavy algorithmic stuff.

Speaking of  algorithmic stuff: I don't know if you do this already, but if you choose the sampling window right (presumably exactly a low octave), you could make use of aliasing in the DFT and need fewer sampling points, a.k.a RAM.

cnlohr, 2016-08-19:
Ok, true... but janitors at CERN don't explain to the scientists how to speed up the tests by 60%

The data actually is run into the dft on the fly, so there's no true sampling window.  It uses an IIR to fall the signal off.  That way basically no ram is needed other than the current sample's real and imagine filtered components.

cnlohr, 2016-08-20:
Random other question - I can't find any other form of contact info.  Would you be up for shooting me an email so I have direct contact?

con-f-use, 2016-08-20:
You're right, a proper introduction might be in order. Hi, my name is Chris, I just send you an E-Mail. Good to meet 'ya. :-D

con-f-use, 2016-08-21:
@cnlohr did you get my E-Mail? Just asking because spam filters are aggressive these days.

cnlohr, 2016-08-21:
I did get your email.  This issue remains an issue though, Mr. "I just re-architected the last 2 years Charles' messes"

con-f-use, 2016-08-21:
Damn, I was hoping, you added me as a collaborator, so I could close this ;-)

cnlohr, 2016-08-22:
Ah!  I found out how to add collaborators.  Now, if only I could figure out how to lock you from closing this issue.

closed 2016-08-22 by con-f-use