cnlohr/esp82xx #5: Improve mechanism for command parsing

cnlohr, 2016-08-19:
Right now, we are using 	things like:

```
char * des = nam ? (char *)ets_strstr( (char*)(nam+1), "\t" ) : 0;
```

which is just awful to be used so many places. We should write our own, perhaps where it also re-writes the byte to be a null and advances the pointer by reference?

i.e.

```char * CaptureAndAdvance( char ** data );```

Called in the following way:

```
    char * nam = CaptureAndAdvance( &pusrdata );
    char * des = CaptureAndAdvance( &pusrdata );
```

This will produce much smaller code and I think it's easier to read.

cnlohr, 2016-08-22:
Maybe another parameter, "ERR" or something?

cnlohr, 2016-08-23:
Ok.  This is complete. 

```
char * parameters;  <<set global variable before using parameter popping
uint8_t  paramcount; <<Set this to 0 before doing parameter popping
int MyNum = ParamCaptureAndAdvanceInt();
int MyVar = ParamCaptureAndAdvance();
```

con-f-use, 2016-08-23:
> /tmp/ccnbGE4K.o (symbol from plugin):(.text+0x0): first defined here
esp82xx/fwsrc/esp82xxutil.c:281:40: error: 'memchr' has already been defined
 const ICACHE_FLASH_ATTR unsigned char *memchr(const unsigned char *s, int c, size_t n)
                                        ^
esp82xx/fwsrc/esp82xxutil.c:281:40: note: previously defined here
 const ICACHE_FLASH_ATTR unsigned char *memchr(const unsigned char *s, int c, size_t n)
                                        ^
esp82xx/fwsrc/esp82xxutil.c:217:26: error: 'strdup' has already been defined
 char * ICACHE_FLASH_ATTR strdup( const char * src )
                          ^
esp82xx/fwsrc/esp82xxutil.c:217:26: note: previously defined here
 char * ICACHE_FLASH_ATTR strdup( const char * src )
                          ^
lto1: fatal error: errors during merging of translation units
compilation terminated.

You seem to have redefined a few function, that were already defined in the SDK...

closed 2016-08-23 by con-f-use