cnlohr/esp82xx #9: Makefile automatically pull recursive

cnlohr, 2016-08-19:
@con-f-use  This is actually an issue for other projects, but is there some easy way of adding a line to the Makefile of other projects so they can automatically detect that they were git pulled without proper --recursive flags and then actually switch over and pull esp82xx?

This would be a change to the child projects, not this project, but would be very useful since people probably aren't used to git pull --recursive.

con-f-use, 2016-08-19:
I'd put it in `Makefile.example` of esp82xx as well as in the top-level projects' main Makefiles. I don't have the time to test it right now, but I'm thinking of a target like that:

```
getsubmods : esp82xx/LICENSE
    git submodule update --init --recursive
```
And than rewrite the dependencies of `$(TARGET) : $(SRCS) Makefile` into:
```
$(TARGET) : getsubmods $(SRCS) Makefile
```

cnlohr, 2016-08-19:
So, I was jump-happy and committed my small change in the projects now, but after reading this, I am having second thoughts.  Leaving this issue open and will look into this later today to swap it around to your way.

con-f-use, 2016-08-19:
Well that would only update and already puled submodule. Since it's in the submodule, we can't use it in top level-project, when the user forgot to `git clone --recursive`

closed 2016-08-19 by cnlohr

cnlohr, 2016-08-19:
So... Why not do it like this:

```
-include esp82xx/common.mf
-include esp82xx/main.mf

ifndef TARGET
$(info Modules were not checked out... use git close --recursive in the future. Pulling now.)
$(shell git submodule update --init --recursive)
endif
```

con-f-use, 2016-08-19:
I tried something similar first. Make complained and refused to execute. Forgot why exactly. Maybe at least one rule has to be present or something.

cnlohr, 2016-08-20:
I don't know what version of Make you're using...  I just tried 'make burn' with mine, and it did pull a new copy. 

```
Modules were not checked out... use git close --recursive in the future.
Pulling now.
Cloning into 'embedded8266/esp82xx'...
remote: Counting objects: 119, done.
remote: Compressing objects: 100% (85/85), done.
remote: Total 119 (delta 51), reused 97 (delta 30), pack-reused 0
Receiving objects: 100% (119/119), 95.24 KiB | 0 bytes/s, done.
Resolving deltas: 100% (51/51), done.
Checking connectivity... done.
Makefile:8: *** multiple target patterns.  Stop.
```

I don't think it's necessary to define additional rules?

```
ifndef TARGET
$(info Modules were not checked out... use git close --recursive in the future.)
$(info Pulling now.)
$(shell git submodule update --init --recursive)
endif
```

Wow, I can't believe we're beating this "nice to have" so hard :(

cnlohr, 2016-08-20:
Let's just call this discussion closed.  Everything is fine now.

commit 8d5d6580 (2016-08-19): Merge pull request #11 from con-f-use/master

Closes #9: Automatically get submodules also re-enable shell sdk variable, various readme stuff