cnlohr/esp82xx #13: Make sure websockets does full binary communication!

cnlohr, 2016-08-20:


con-f-use, 2017-05-15:
What? Also old...

cnlohr, 2017-05-15:
hhhnnnggggg  I keep pushing this off, though it would be REALLY helpful.  There are bits that need to be set in the responses as well as a change in the javascript.  It would be super cool to be able to just send binary data instead of needing to hex-encode large binary chunks.

con-f-use, 2017-05-15:
I think I did something like this already somewhere in the code. Might be in on of my projects not on git. Just didn't get what you mean at first.

cnlohr, 2017-05-15:
Ah! Yes.  Hmm, I SHOULD have tried it last night instead of jumping directly to using hex encoding.

con-f-use, 2017-05-16:
Found it. It was in the led [color setting web-interface](https://github.com/cnlohr/esp8266ws2812i2s/blob/master/web/page/main.js#L80) for [esp8266ws2812i2c](https://github.com/cnlohr/esp8266ws2812i2s).

cnlohr, 2017-05-16:
That's still using hexpairs.

con-f-use, 2017-05-16:
Well the data `qDat` is an array of 8 byte `ints`. Is not converted or decoded to/from hex, i.e. it's not only using the 0-9 & a-f. Not sure what else you mean.

cnlohr, 2017-05-17:
Hmm... I think it is anywhere zero and reserved chars may be used.

http.c. line 643.  The opcode of the websockets packet is still 1.

cnlohr, 2017-05-17:
I think I just restricted the possible values to preclude it from hitting 0.

cnlohr, 2017-05-17:
Ah! Ok, so changing that kinda makes things... work... ish... I'm going to try to change it in the backend javascript maybe to autodetect?

cnlohr, 2017-05-17:
no need... but now I am once again battling the git.

commit b4777345 (2017-05-17): Closes #13 - Add support by switching protocol to binary and providing most users in Javascript space the same string, though the binary blob is also now available.
commit 51c38dec (2017-05-20): Add init-data, improve readme and fix minor bugs (#57)

* Add features to allow for full flash erase and provide init_data for both regular ADC and ADC33. Also, cleanup location of some functions in main.

* Closes #13 - Add support by switching protocol to binary and providing most users in Javascript space the same string, though the binary blob is also now available.

* USe esp82xx version if top-level not a git project

* Document init-data in README.md

* Minor typos in README.md - Thanks to @bbkiwi
closed 2017-05-17 by cnlohr