cnlohr/esp82xx #19: Order of inclusion

cnlohr, 2016-08-23:
The top of the makefiles read:

```
include user.cfg
-include esp82xx/common.mf
-include esp82xx/main.mf
```

Shouldn't user be after common, that way it can override things like SDK?

con-f-use, 2016-08-23:
Nope, `common.mf` needs some of the variables in `user.cfg`.

cnlohr, 2016-08-23:
How can they override things like SDK then?  Or other variables within it.  I guess I am just having trouble specifying specific Espressif SDK versions.

con-f-use, 2016-08-23:
> How can they override things like SDK then?

One would just change the value of `DEFAULT_SDK` in `./user.cfg`.

The default value is

    SDK_DEFAULT  = $(HOME)/esp8266/esp-open-sdk 

Change that to 

    SDK_DEFAULT  = $(HOME)/esp8266/esp-open-sdk2.0.0

if you want to use the 2.0.0 version. At least if they installed the SDK with the @pfalcon's Makefile, because that has a softlink `sdk -> esp_iot_sdk_v2.0.0/` in its root directory.

For a one-off build, just to see if it compiles with another SDK, you can also pass the variable to Make as an argument:

    make all ESP_ROOT=~/esp8266/esp-open-sdk2.0.0

cnlohr, 2016-08-23:
So, the big deal here isn't the location of the pfalcon esp-open-sdk, it's the location of the espressif esp-nonos-sdk... Which I think you know, sorry, the fact they're both SDKs is hard.

Anyway, the problem isn't even which version someone wants to use, which yes, it would absolutely make sense to do in shell variables, or make arguments.

The problem is that /some/ projects will be dependent on specific versions of the SDK.  For instance, until last night, colorchord was incompatible with 1.5.4, so I had to specify a specific version.  So, things that would be done per-user or per-environment are right out.  Per-project is the only way to go.

con-f-use, 2016-08-23:
Then the top-level Makefile would be the best place.

cnlohr, 2016-08-23:
Where in the makefile?  Hmm, sorry, you just gave such a beautiful explanation up there... I guess where would be the idea place in a Makefile?

con-f-use, 2016-08-23:
Np. See my [edit](https://github.com/cnlohr/esp82xx/issues/19#issuecomment-241799138).

cnlohr, 2016-08-24:
Hmm, that totally doesn't work.  It tries pulling GCC and all its friends from there, too.

con-f-use, 2016-08-24:
Now I'm con-f-used. Here is my setup:

I have different versions of the SDK (all build from @pfalcons Makefile) at:
 - `/home/confus/build/esp-open-sdk1.5.0`
 - `/home/confus/build/esp-open-sdk1.5.2`
 - `/home/confus/build/esp-open-sdk1.5.4`
 - `/home/confus/build/esp-open-sdk2.0.0`
 - `/home/confus/build/esp-open-sdk -> `/home/confus/build/esp-open-sdk1.5.0` softlink to the version that usually works best

In my `.bashrc` I have `export ESP_ROOT=/home/confus/build/esp-open-sdk`

I test if stuff compiles with on other SDK version (e.g. `2.0.0`) with:

    make burn ESP_ROOT=/home/confus/build/esp-open-sdk2.0.0

What is your setup and what exactly do you want to do?

cnlohr, 2016-08-24:
I want the GCC that comes with esp-open-sdk on all of my projects.  There is no need to tie the Espressif SDK to any esp-open-sdk version, as they are completely independent.

I want to be able to select a specific Espressif NonOS SDK version separate from the esp-open-sdk.

I really wish we had better words for these things.

con-f-use, 2016-08-24:
Ah, now I got you. Have you tried redefining the following stuff in the top-level `Makefile`?

```
GCC_FOLDER = $(ESP_ROOT)/xtensa-lx106-elf
ESPTOOL_PY = $(ESP_ROOT)/esptool/esptool.py
SDK        = $(ESP_ROOT)/sdk

XTLIB        = $(SDK)/lib
XTGCCLIB     = $(GCC_FOLDER)/lib/gcc/xtensa-lx106-elf/$(ESP_GCC_VERS)/libgcc.a
FOLDERPREFIX = $(GCC_FOLDER)/bin
PREFIX       = $(FOLDERPREFIX)/xtensa-lx106-elf-
CC           = $(PREFIX)gcc
LD           = $(PREFIX)ld
AR           = $(PREFIX)ar
CP = cp
```

cnlohr, 2016-08-24:
But, but... but.....  I am asking you if I wanted to override that one specific component, what would I make my Makefile, or user.cfg look like?

con-f-use, 2016-08-24:
```
# In Top-Level Makefile
include user.cfg
-include esp82xx/common.mf
-include eps82xx/main.mf

SRCS += whatever.c \
  stuff.S

SDK = ~/path/to/NonOS/sdk
```
Maybe? Didn't try it.

cnlohr, 2016-08-24:
I must be going crazy.  I though I tried exactly that, but I guess I didn't.  Indeed.  This is precisely the thing I wanted.  I'm really sorry, thanks for dealing with my slowness on these issues.  I thought last time I tried it it was trying to pull the compiler from there.

closed 2016-08-24 by cnlohr

con-f-use, 2016-08-24:
No worries. Believe me, I've been there. We all have.