cnlohr/esp82xx #23: Custom command Cmd_WiFi "WR" not working on web

brainstorm, 2016-08-24:
Hi @cnlohr,

Sorry to bother you with this, but I was about to write a method to return a wireless scan as an array, and voila, there you have already a `case` in cmd_WiFi.

But I tried to issue the commands `WR1`, `WR`... to no avail, wonder if it really works via the WebGUI?

cnlohr, 2016-08-24:
I just changed the way it worked last night so I could free up about 500 bytes of RAM.  I tested it and it worked a-okay.  Are you not seeing positive behavior from the web GUI?

WS first, wait for comms to come back, then WR

brainstorm, 2016-08-25:
Just tried what you mentioned (WS, wait for comms and then WR) and I just get a `!WR` back :-S

[image]

cnlohr, 2016-08-25:
Aaha!  I see a list of APs above there. So, the webpage, itself polls for WR frequently.  Whenever you issue a WS, the device disconnects, scans, and keeps a copy of the scan list around.  Once reconnected, the webpage pulls that list with "WR." Once received, it erases the list on the ESP and boom.  Can't see it anymore.

You might be able to get away with this by closing the "Wifi" tab.

brainstorm, 2016-08-25:
Yeah, it was exactly that, now I get that array/list back, thanks! ;)

closed 2016-08-25 by brainstorm