cnlohr/esp82xx #34: image.elf section `.irom0.text' will not fit in region `irom0_0_seg'

Def152, 2016-08-30:
I'm using ESP8266-12E. I needed to make a http get request so i found [httpclient](https://github.com/Caerbannog/esphttpclient) code. I fixed some warnings but then I got this:

```/home/karol/git/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: image.elf section `.irom0.text' will not fit in region `irom0_0_seg'
/home/karol/git/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: region `irom0_0_seg' overflowed by 6576 bytes```

How can I fix this? As far I as know you modified a bit memory layout. We are using 32Mb flash but in makefile there is a -fs 8m flag. I'm also curious about OTA. Could you explaing a little bit how it is different from espressif way?

con-f-use, 2016-08-30:
Which version of the SDK do you use? It should be <=1.5.2

Def152, 2016-08-30:
I'm using 1.5.4, I will try with 1.5.2

con-f-use, 2016-08-30:
You can try, but I'd guess, the extra code for httpclient doesn't fit in the IRAM of the esp.

Def152, 2016-08-30:
This is a noob question but the httpclient code is running from flash. There is a ICACHE_FLASH_ATTR. How can I run out of IRAM?

cnlohr, 2016-08-30:
We need to add the /really/ fast ADC function to esp82xx.  I tried out the 1.5.4 one and it wasn't sufficient for handling regular audio processing from what I remember.

The code is split into "IRAM" which is loaded into the core, and is super duper fast.  The rest is loaded into IROM which stays on the flash chip, but is cached, so it's still pretty quick.  One of the biggest problems is Espressif for some reason decided to start gobbling up the precious IRAM.  Whenever handling interrupts or modifying flash code, that code needs to (unfortunately) live in IRAM.  

Sticking ICACHE_FLASH_ATTR in front of the function names of functions will place them into IROM and free up the IRAM :)

Def152, 2016-08-30:
Ok I a bit lost. I used ICACHE_FLASH_ATTR on every function httpclient is using and it needs IROM not IRAM, If I edit eagle.app.v6.ld irom0_0_seg from `org = 0x40240000, len = 0x3C000` to `org = 0x40210000, len = 0x6C000` it compiles but then esp8266 crashes. I just need more IROM

cnlohr, 2016-09-04:
The bigger problems is the ESP strictly has no more than 40kB IRAM.  The Espressif SDK was compiled for use in IRAM.

Def152, 2016-09-04:
Correct me if I'm wrong. If function has ICACHE_FLASH_ATTR it goes to SPI Flash (IROM). Otherwise it is placed in IRAM section (also on SPI Flash but it is copied at the beginning to embedded esp memory). I know that IRAM is full of their SDK things and there is no space for my own code. But there is still one thing that I dont get. All the functions in httpclient.c are with ICACHE_FLASH_ATTR, so how is it possible to run out of IRAM?

cnlohr, 2016-09-05:
So, just checking:  what version of the SDK are you using now?  Older versions have tons of free space.

Def152, 2016-09-06:
On 1.5.2 I still need 4212 bytes, on 1.5.4 i needed 6576 bytes.

cnlohr, 2016-09-06:
Something is definitely going on here with where your functions and/or data is residing.  There should be a lot more free room than that on 1.5.2.  Can you c&p the build output?

Def152, 2016-09-07:
Here it is:
`karol@karol-VB ~/git/esp $ make burn`
```esp82xx/common.mf:28: Warning: No shell variable 'ESP_ROOT', using '/home/karol/esp8266/esp-open-sdk'
fatal: bad revision 'HEAD'
/home/karol/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I/home/karol/esp8266/esp-open-sdk/sdk/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DVERSSTR='"Version:  - Build śro, 7 wrz 2016, 23:09:31 CEST with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878 "' esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c user/httpclient.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L/home/karol/esp8266/esp-open-sdk/sdk/lib -L/home/karol/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a /home/karol/esp8266/esp-open-sdk/sdk/lib/libmain.a /home/karol/esp8266/esp-open-sdk/sdk/lib/liblwip.a /home/karol/esp8266/esp-open-sdk/sdk/lib/libssl.a /home/karol/esp8266/esp-open-sdk/sdk/lib/libupgrade.a /home/karol/esp8266/esp-open-sdk/sdk/lib/libnet80211.a /home/karol/esp8266/esp-open-sdk/sdk/lib/libwpa.a /home/karol/esp8266/esp-open-sdk/sdk/lib/libphy.a /home/karol/esp8266/esp-open-sdk/sdk/lib/liblwip.a /home/karol/esp8266/esp-open-sdk/sdk/lib/libpp.a /home/karol/esp8266/esp-open-sdk/sdk/lib/libcrypto.a /home/karol/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T /home/karol/esp8266/esp-open-sdk/sdk/ld/eagle.app.v6.ld -B/home/karol/esp8266/esp-open-sdk/sdk/lib -o image.elf
/home/karol/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: image.elf section `.irom0.text' will not fit in region `irom0_0_seg'
/home/karol/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: region `irom0_0_seg' overflowed by 4212 bytes
collect2: error: ld returned 1 exit status
esp82xx/main.mf:72: polecenia dla obiektu 'image.elf' nie powiodły się
make: *** [image.elf] Błąd 1```

cnlohr, 2016-09-08:
Those bums.  Looks like they just bumped everything to 2.0.  You will need to use 1.5.2.  Though I am still very surprised by how MUCH it's overflowing.  Most of my projects will still fit in 2.0, but just barely.

I was looking for ```-DICACHE_FLASH``` which is in there. Can you post your user/httpclient.c?

Def152, 2016-09-08:
https://gist.github.com/Def152/2d8ec396cf8adab7705e0bac2e26fb82

cnlohr, 2016-09-08:
Ok, that is very odd, it should work.  Just wondering do you have this in a forked repo that I can try building?

Def152, 2016-09-08:
Here is repo: https://github.com/Def152/esp1 , I followed [this](https://github.com/cnlohr/esp82xx#usage)

cnlohr, 2016-09-09:
Well, this is unfortunate.  It appears ```espconn_gethostbyname```  poisons the whole thing.  @con-f-use  do you have any idea why inclusion of this function increases the size of the .iram (.text) area by what seems to be almost 6kB?