cnlohr/esp82xx #42: Cannot make any changes in website

ArchontPL, 2016-10-25:
Hi,
Finally I was able to compile esp-open-sdk, esp82XX-basic and burn .bin to my ESP board. I try to change `user.cfg` file (`PAGE_TITLE = esp82xx-basic` to `PAGE_TITLE = esp82xx-basic-mod`), then `make purge`, `make all` and `make burn`. All process was succesfull, but even when I refresh site on browser I cannot see any changes in page title. What I'm doing wrong?

con-f-use, 2016-10-26:
Judging from the other issues you filed and the fact, that you are on cygwin (Windows), I'd say either grep or sed are not working properly. `PAGE_TITLE` is substituted into `./web/page/index.html` by sed. I cannot reproduce this issue on my Ubuntu.

ArchontPL, 2016-10-26:
@con-f-use no, to build SDK I had to manually install packages (AFAIK Cygwin don't have package manager like apt-get). To do that I cloned SDK using `git clone --recursive https://github.com/pfalcon/esp-open-sdk.git` then `make STANDALONE=n`. 
I check in home if sed is working properly.

closed 2016-10-26 by ArchontPL