cnlohr/esp82xx #44: can't burn web page due to missing usb library

pwilga, 2017-02-05:
Hello,

I have followed instructions in README file. Code compiles without any error. I was able to flash firmware via make burn command. Nevertheless I'm getting following error during web content flashing. I coludn't even find a place where this usb-1.0 lib is included. After two days I'm out of ideas - I tried to run exp82xx-basic project but it ends up with the same error message. Thank you in advance

make burnweb

rm -f mfsmaker page.mpfs pushtodev execute_reflash tmp/*
gcc -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50 mfsmaker.c -o mfsmaker  -lusb-1.0
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [mfsmaker] Error 1
make: *** [burnweb] Error 2

con-f-use, 2017-02-05:
`libusb` is an external library. You need to install it via the usual means of your operating system. Under Ubuntu that would be:

    sudo apt-get install libusb-1.0-0-dev

closed 2017-02-05 by con-f-use

pwilga, 2017-02-05:
Hello con-f-use,
 
I'm working on Mac, but I will find way to install this library :)
Much appreciated !

con-f-use, 2017-02-05:
In that case use [macports](https://www.macports.org/): `sudo port install libusb`

pwilga, 2017-02-06:
Hello,

Thank you for your support. I still have couple of issues, but I'm not sure if it's lack of knowledge, sdk or sth else. 

After your remark everything compiles ok, and works fine ;) The only difference is that I'm using brew instead of port. For Yosemite just do:
brew install libusb
and
brew install nmap
to get 'make getips' working :)

pwilga, 2017-02-06:
forgot to add 👍 good job