cnlohr/esp82xx #45: Incude SDK and possible the whole toolchain

con-f-use, 2017-02-08:
The purpose of this issue is to discuss an e-mail conversation, I had with @cnlohr :

> Espressif has sent me a patchwork of various updates to their SDK.  I am adding them to https://github.com/cnlohr/esp_nonos_sdk

> One big benefit is that it seems with this and the custom linker script, the IRAM issues are a thing of the past.  What do you think it would take to make esp82xx also check out the new SDK?

My response in a nutshell:
> You mean as in a Make target for cloning/donwloading the SDK [...]?
> To answer your question, I guess it wouldn't take much to get it running [...]. 
> Submodule `git@github.com:cnlohr/esp_nonos_sdk.git` into esp82xx [and adapt the make file].

> I've thought about [including the whole Toolchain] at one point. 
> Would be an enormous relief for new users. 
> The reason why I didn't, was that it's probably only going to work for a few users that happen to build on a similarly configured OS.

So the SDK part should just be a matter of small makefile changes and a submodule.

For getting and compiling the whole toolchain, there's already [my Ubuntu Script](https://gist.github.com/con-f-use/d086ca941c2c80fbde6d8996b8a50761) that uses [pfalcon's Makefile](https://github.com/pfalcon/esp-open-sdk). We could just include that, modify it to use @cnlohr 's patched SDK and call it a day. That would ignore other non-Ubuntu based systems completely.

con-f-use, 2017-02-09:
I have forked @pfalcon's [esp_open_sdk Makefile](https://github.com/pfalcon/esp-open-sdk) and modified it so it uses [your version the SDK](https://github.com/cnlohr/esp_nonos_sdk). Here is my fork: https://github.com/con-f-use/esp-open-sdk

Can @cnlohr  test it please?

con-f-use, 2017-02-09:
> We are gonna have to come to terms with this git@ thing. If you use the oauth tokens with github, it doesn't have to prompt you for password or anything, either.

Never heard of "oauth tokens" but I agree, that neither HTTPS nor SSH are the solution that fits all. If I had my wish, git would just automatically try everything until something works.

con-f-use, 2017-02-10:
Btw. where exactly did you run into trouble with the `@git` thing?

cnlohr, 2017-02-13:
gaaaahhh I have to push this off another day!

con-f-use, 2017-02-16:
Btw. when I use your version of the SDK, I get:
```
/home/jan/Desktop/data/tmp/esp-open-sdk/sdk/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x6d8): undefined reference to `user_rf_cal_sector_set'
/home/jan/Desktop/data/tmp/esp-open-sdk/sdk/lib/libmain.a(app_main.o): In function `flash_data_check':
(.irom0.text+0x847): undefined reference to `user_rf_cal_sector_set'
```

cnlohr, 2017-02-16:
With which project?  For most of my projects, I've c&p'd the example espressif code for those functions.  I may have missed a few.  That feature was added after 1.5.2, I believe.

con-f-use, 2017-02-16:
As I said, the example files that currently come with esp82xx.

con-f-use, 2017-02-16:
Only remaining issue:

> [...] would it be enough to submodule your sdk-version into esp82xx and then set the appropriate path variables in the make files (e.g. SDK and XTLIB in ./esp82xx/common.mf)? I don't fully understand how the SDK integrates into the other tools and whether something form the SDK needs to be compiled client side.

cnlohr, 2017-02-18:
For easy use across Windows/Linux/etc. And for integration to the github auth storing thing, I'm really finding HTTPS is better than SSH.  Maybe we should include instructions for both whenever working with it?

con-f-use, 2017-02-19:
I'd stick with the HTTPS simply because it's more portable. I will probably write a small note about how to setup gnome's keyring for Ubuntu to work with it. It was not as straight forward as it should have been. Still I'd like to know where you run into trouble, so we can fix that. An inconsistent solution is most annoying.

cnlohr, 2017-02-19:
I'm just saying that in contrast to the ssh you seem to prefer.  Though, github is suggesting a move away from stored passwords and toward keys.

con-f-use, 2017-03-03:
Okay, I think I've included your version of the sdk now and got a decent mechanism to [start new esp82xx projects](https://github.com/cnlohr/esp82xx/tree/ft-include_sdk#start-a-new-project). 

It's all in the [ft-include_sdk](https://github.com/cnlohr/esp82xx/commits/ft-include_sdk) branch. Please check that out and test it. If @cnlohr is happy with it, we can merge it into dev and then squash-merge it into master.

cnlohr, 2017-03-15:
Asked elsewhere, just double-checking is the case sensitive?

con-f-use, 2017-03-15:
?

cnlohr, 2017-03-15:
"CNLohr/esp82XX" but yes.  This seems very cool!  Please push!

closed 2017-05-13 by con-f-use

cnlohr, 2017-05-13:
This thread has been so drawn out over so long, was it actually done as per your satisfaction?