cnlohr/esp82xx #47: Include other esp related libraries

con-f-use, 2017-03-08:
When issue #45 is done, I think the code for [ws2812](https://github.com/cnlohr/esp8266ws2812i2s), [usb](https://github.com
/cnlohr/espusb) and [ethernet](https://github.com/cnlohr/espthernet) could be included in esp82xx.

cnlohr, 2017-03-15:
What do you mean by "included in" 

Also, is github not case-sensitive when it comes to paths and projects?

con-f-use, 2017-03-15:
I mean, once could make a generalized library out of it and add that as optional part.

Why are you asking about case sensitivity? Did I miss some upper-lower-case stuff somewhere?

cnlohr, 2017-03-15:
Though that is possible, I really don't understand how I'd do it.  The ESP is very limited and all of those projects are very large.  Wouldn't be able to support multiple projects per ESP.