cnlohr/esp82xx #52: unfubar strcat() in user_main.c

geekmaster, 2017-05-13:
"that's really wrong.  File a bug report.
that's like /really/ wrong.
It's because the standard library is missing strcat, I believe."
-- cnlohr

con-f-use, 2017-05-13:
Yeah, it seems some magic code, I didn't write has injected itself into the last merge. No idea what happened there. Also the rf_cal functions and so on changed.

cnlohr, 2017-05-13:
I vaguely recall putting in the strcat code.  Though I feel completely idiotic about writing it now.  I don't know what I was thinking.

charrx I think is when data comes back down the serial line.  That way you can easily hook incoming serial bytes.

cnlohr, 2017-05-13:
P.S. Please use your code if you've tested it.  I also noticed 'man cat' has an implementation:

```
           char *
           strncat(char *dest, const char *src, size_t n)
           {
               size_t dest_len = strlen(dest);
               size_t i;

               for (i = 0 ; i < n && src[i] != '\0' ; i++)
                   dest[dest_len + i] = src[i];
               dest[dest_len + i] = '\0';

               return dest;
           }
```

con-f-use, 2017-05-14:
Fixed in 93074f1

closed 2017-05-14 by con-f-use

cnlohr, 2017-05-15:
and it works :-D