cnlohr/esp82xx #53: Random charactes on serial with ESP not booting

con-f-use, 2017-05-15:
With the latest version of esp82xx I'm getting something strange on fresh esp-12-E modules

After
```
git clone --recursive https://github.com/cnlohr/esp82xx.git
cp esp82xx/Makefile.example Makefile
make project
make burn
make burnweb
```
I only get an endless sequence of seemingly random character sequences on the UART and the ESP doesn't connect to WiFi or respond to anything but putting it in the bootloader. When I flash an old version and then the new version again and set WiFi credentials, everything works. Did @cnlohr experience something similar with the latest commit?

P.S. I tried multiple ESPs with different programmes.

cnlohr, 2017-05-15:
I don't believe so.  Or, at least I JUST started using it on esp8266lighthouse, you may want to try checking that out and seeing if it endless-crashes for you.  One thing I find dubious is the continuous flashing of the initdata.  https://github.com/cnlohr/esp8266lighthouse

cnlohr, 2017-05-15:
I have, however, noticed this with some other hardware.  I thought the hardware was the issue.  I may be wrong.

con-f-use, 2017-05-15:
At first I thought it hat something to do with loading the WiFi credentials. Something read from a virgin flash with random data. But I don't think so. The only code stuff we have changed was a) the sdk b) the rf_cal routine c) strcat. Or can you think of more changes, we made recently?

> One thing I find dubious is the continuous flashing of the initdata.

Not sure what you mean here.

cnlohr, 2017-05-15:
dubious = I was just working with the project "esp8266lighthouse" last night and I believe using the newest esp82xx

con-f-use, 2017-05-15:
Yes you were, I saw the lifestream, but missed, what you describe as "continuous flashing of the initdata". Talking of continous flashing, the led blinks like crazy (constantly resetting?) when this bug occurs.

cnlohr, 2017-05-15:
I have definitely seen it.  I will try to make time to burn with stock firmware tonight.  Can you try burning the esp8266lighthouse code and see if you constantly reset?

con-f-use, 2017-05-15:
I don't have any "fresh" modules with the factory-default AT-firmware on them anymore. I've only seen the bug on them. So unfortunately no.

cnlohr, 2017-05-15:
Ok, now I am reallllyy confused by what you're saying.  You're saying you have busted modules that seem unable to work EVER or you do something and they work fine?

con-f-use, 2017-05-15:
I'm saying:

1.) I have a new module. It is perfectly fine and has (by factory) the AT firmware on them
2.) I flash the latest version of esp82xx to them
3.) After step 2. the module is caught in an endless loop of resets, it is now "faulty".
4.) I flash an older version of esp82xx to the "faulty" module and set it up to connect to my home's WiFi.
5.) The formally "faulty" module now works perfectly with either the old or (if I flash it with the latest) new firmware version. The bug forever disappears.

cnlohr, 2017-05-16:
I just pushed to dev a fix that should help with this.  I think something may have been going on fishy with the init data.  I'm not sure.  Make sure you can erase chip and install the init data.

Also, should I not have pushed that to dev?

I am a little confused as to how main and dev interact.

cnlohr, 2017-05-16:
Oooohh nooooo I've been using and flashing the master branch this whole time and never the dev branch!!!  That could explain why I haven't seen it.

So here's what I want you to do.  Can you dump the FULL flash off a fresh-from-factory one?  

```
esptool.py -b 1000000 read_flash 0 4194304 factory.bin
```

It'll take a bit, even at 1

cnlohr, 2017-05-16:
Everything's broken again.  In my project where I have esp82xx, I can't get the newest dev, and if I try git pull origin dev, I get this 

```
From https://github.com/cnlohr/esp82xx
 * branch            dev        -> FETCH_HEAD
Auto-merging user.cfg.example
CONFLICT (content): Merge conflict in user.cfg.example
Auto-merging main.mf
CONFLICT (content): Merge conflict in main.mf
Auto-merging fwsrc/user/user_main.c
CONFLICT (add/add): Merge conflict in fwsrc/user/user_main.c
Auto-merging README.md
CONFLICT (content): Merge conflict in README.md
Automatic merge failed; fix conflicts and then commit the result.
```

I promise I am trying.  Man, git is hard.  And with multiple mainline branches it's even harder :(.

cnlohr, 2017-05-16:
It's all busted, no matter what I revert, ```git checkout origin dev``` just hoses the tree.

cnlohr, 2017-05-16:
I neeeeddd to sleeeeepppp but it's sooooo broken

con-f-use, 2017-05-16:
Okay so I cleaned up. Just reverted all your changes save the last one and leveld the `master` and `dev` branches. 

You will need a fresh clone of the dev branch, if you want to make further changes:

    git clone --recursive https://github.com/cnlohr/esp82xx.git -b dev

I think `dev` is the place to experiment and fix. Once `dev` is working, we squash-merge into master.

cnlohr, 2017-05-16:
Are you okay with those changes I made?  I.e. the moving around?

Did you get a chance to dump a fresh build?

Have you tried erase and the init flash?

Charles

con-f-use, 2017-05-17:
The moving around is fine. As to the other points: I'll have to get to them later.

cnlohr, 2017-05-17:
So, the tip-off to me was the following:

*If you erase the chip and try running it, no matter what you do, it will always get stuck in a reboot loop.* At least until you flash init data.

That leads me to believe that the issues may be related. Perhaps the init data somehow changed between SDK versions?

con-f-use, 2017-05-18:
Okay init-data seems to be a component. Still having some reboot problems after flashing the init-data but at least I see something on the UART and resetting is less frequent. Might also be, that my board is wonky (on top of things). Will need further testing before closing the issue. Just wanted to mention it.

cnlohr, 2017-05-18:
There's so many things that can go wrong.  I am glad that overall, these systems are slowly paying down their technical debt :)

Glad you asked!

One lets you use the external ADC as a regular ADC.  The other lets you monitor the voltage of the 3.3v rail.  It's really useful for things like con badges, etc.

cnlohr, 2017-05-20:
I see you closed it but without comment?  Did you determine if this was indeed the fix?  Erase-and-initdata?

con-f-use, 2017-05-20:
Oops, no must have hit the button by accident. Still waiting for my new boards and modules to come in.

cnlohr, 2017-05-20:
gotcha

cnlohr, 2017-05-23:
So, just to point it out.  I got a chip that was stuck in constant reboot land.  Did the erase and initdefault and it fixed it.

con-f-use, 2017-05-30:
I got my new boards, chips and modules. Everything werks ze fine! 'tis fixed.

PR #57 (2017-05-20): Add init-data, improve readme and fix minor bugs
 - Make target `initdefault and `init3v3` hopefully fix #53. Former uses the ADC as normal, latter uses it to monitor Vcc
 - Use esp82xx version if not in a git project 
- Flash erase make target
 - Many README.md fixes
closed 2017-05-30 by con-f-use