cnlohr/esp82xx #54: Reboots with default settings after toggling gpios

Damme, 2017-05-16:
When setting wifi settings in web-gui, the "change settings" button does not save settings to flash, I have to send command "Is" to save settings.

My main problem is that settings reverts to default then device is crashing by unknown reason. One way to reproduce this is to randomly toggle gpio pins via web-gui. This is the log from then it reboots to factory defaults after crash.

Web-GUI
Version: unknown - Build Mon, 15 May 2017 20:54:44 +0200 with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=5Loading Settings: af / 0 / 69 / 69
Settings Loaded: ESP_D60606 / Default
Opmode: 1
Station mode: "Q-2.4":"xxxxxxxx" (bssid_set:0)
Boot Ok.
IGMP Joining: cc282a0a fb0000e0
STAT: 5
IP: 10.42.40.204
NM: 255.255.252.0
GW: 10.42.41.200
WCFG: /Q-2.4/xxxxxxxx/
IGMP Joining: cc282a0a fb0000e0
                             
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30368, room 16 
tail 0
chksum 0x6a
load 0x3ffe8000, len 884, room 8 
tail 12
chksum 0x3d
ho 0 tail 12 room 4
load 0x3ffe8380, len 2832, room 12 
tail 4
chksum 0xe6
csum 0xe6
����g�s��n|�$dll`��;�$�d�䤨H����2XX Web-GUI
Version: unknown - Build Mon, 15 May 2017 20:54:44 +0200 with -DICACHE_FLASH -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=5Loading Settings: af / 0 / 69 / 69
Settings Loaded: ESP_D60606 / Default
Opmode: 2
Default SoftAP mode: "ESP_D60606":""
Boot Ok.
IGMP Joining: 0104a8c0 fb0000e0

con-f-use, 2017-05-16:
@cnlohr is fixing #53 at the moment, which might be related.

Damme, 2017-05-16:
Tested with both Master and Dev.

I have successfully changed all gpios (that are changeable to out) without reboot, but then randomly pressing then in fast succession crash happens. Its not one same gpio that toggles the crash but random.

con-f-use, 2017-05-16:
Okay that is strange. As I said in #53, I had endless reboots after flashing the firmware. They seem to be related to the init-data, but random resets on toggling the GPIOs seem very strange. Are you sure it is not a hardware issue? Do have adequate capacitance on the power rail close to the module (at least 10 μF)? Is the problem persistent between different ESP-modules? Is it persistent between different versions of the SDK?

Damme, 2017-05-16:
Hello again
It seams not to be to do with switching gpio to 1/0, but direction in/out , and always switching back an input to IN. There is no external hardware and I tried with and without an extra cap.

What buggers me the most though is that the settings reverts to default. My saved settings are not loaded.

cnlohr, 2017-05-16:
That is really weird.  

Also! @Damme it looks like there is a change in the SDKs as of recent.  Try erasing your ESP, and flashing the init data.

```make erase initdefault burn```

Charles

con-f-use, 2017-05-18:
Okay in the mean time, I had one of my modules doing the same thing with resetting on IO direction change. But that went away on it's own. I really thing it's either corrupted flash or a hardware issue. But the bug is real.

Damme, 2017-05-23:
I have put this project to the side at the moment and tried the same module with arduino-esp and seams to work fine.
And yes, I did try erase and init default. I'll come back to this later with a brand new module to see if it was hardware error or not.

cnlohr, 2017-05-23:
Ok.  Just tried with a fresh ESP. ```make erase initdefault``` does in fact fix the constantly rebooting issues.  So, this is likely a different issue.

closed 2017-05-23 by con-f-use