cnlohr/esp82xx #66: Problem with Scan For Wifi

bbkiwi, 2017-06-25:
I've noticed some problems while testing the new cc that uses esp82xx/dev that seem to be related to esp82xx/dev.

1. In AP after scan for wifi (and having not found any) it attempts to return to the AP but fails to establish it. Manual reset will get it going again.
2. In Station mode, sometimes after the scan it forces a reboot and then reconnects back to the station.

[output esp82xx test2.txt](https://github.com/cnlohr/esp82xx/files/1100064/output.esp82xx.test2.txt)
[output esp82xx test.txt](https://github.com/cnlohr/esp82xx/files/1100065/output.esp82xx.test.txt)

cnlohr, 2017-07-19:
So, did that actually fix it?

bbkiwi, 2017-07-20:
I still don't find anything with scan :-O 
But after scanning in AP mode it does reconnect back to AP with no problem.
After scanning in station mode, it no longer reboots.