cnlohr/esp82xx #68: netburn failing

bbkiwi, 2017-07-20:
I'm checking this out. I get !FM4 (MD5 mismatch). 
If I remove -DQUIET_REFLASH it seems to work but gives message 
"Timeout. Good? Maybe?" (a bit of a strange message).

netweb seems to work with either option.

cnlohr, 2017-07-20:
That is /very/ unusual and I have never seen anything like it, but I am also not on the most recent branch, so I would imagine it has to do with the recent changes.  I really can't make much time to commit to this until next week with Artscape coming up and a number of other events, but, I am curious if this happens if you flash the older esp82xx.

bbkiwi, 2017-07-21:
I've tried esp82xx earlier commits 21e23b5 before my changes, and 21e23b5 also. Both give the message "Timeout. Good? Maybe?" and the new code does NOT upload. 
I used 79ce49d and got "Timeout. Good? Maybe?" but the new code DID upload. Also lots of stuff was printed on the serial port (which didn't happen with the two cases where the code did not upload).
[SerialOut.txt](https://github.com/cnlohr/esp82xx/files/1164580/SerialOut.txt)

I tried with cc + esp82xx (most recent on my repo branches devwork and devfixes respl.) got the !FM4 once but mostly get "Timeout. Good? Maybe?" and the new code does NOT upload.  So maybe the !FM4 was just a random glitch there.

cnlohr, 2017-07-21:
oh boy that's weird.  UUggghh I wish I could do this now :(

bbkiwi, 2017-07-22:
Did more tests.  9641825 uploads OK (gives terminal message "Timeout. Good?.." get serial messages A....E and reboots w cause 2)
adb21e5 and 50afc5e upload OK (gives terminal message "Timeout. Good?.." get serial messages A....E  **but** reboots w cause 4 wdt reset)
21e23b5 FAILS (gives terminal message "Timeout. Good?.." **but** serial messages stop at AB and reboots w cause 4 **wdt reset**) but old code is still running afterwards. (I think the wdt reset is triggered in  Cache_Read_Disable() but I can't see how the earlier commits are that much different)
[testnetburn.txt](https://github.com/cnlohr/esp82xx/files/1167090/testnetburn.txt)

bbkiwi, 2017-07-22:
FIXED! Submitted pull.

bbkiwi, 2017-07-22:
Fixed  -DQUIET_REFLASH too. Submitted pull

cnlohr, 2017-07-24:
So... close issue?

bbkiwi, 2017-07-24:
Ok.

closed 2017-07-24 by bbkiwi

cnlohr, 2017-07-25:
Sorry, I just realized that came off the wrong way.  I just wanted to know if everything was golden.  Not badger you for leaving an issue open :(

cnlohr, 2017-07-25:
I really appreciate the effort you're putting in here!

bbkiwi, 2017-07-25:
Not a problem, I didn't even realize your post could be taken as badgering :-)

con-f-use, 2017-07-25:
Just merged everything into master, although #60 and #66 are still open and probably related.
Good work people, ESPecially bbkiwi!