cnlohr/esp82xx #70: Feature request/question

HealV, 2017-07-27:
I don't want to install all this tools on my computer so I have a question. Can I just burn .bin file to the esp?
Thanks

con-f-use, 2017-07-27:
Yes, we could provide a binary to "just burn". I'll get to it this weekend.

HealV, 2017-07-27:
Can you compile whole program for me and send me only bin files and spiffs?
I will then flash it via nodemcu flasher and upload spiffs by arduino.
Thanks again!

cnlohr, 2017-07-28:
That's a little weird of a thing to do for esp82xx, since it doesn't really have a project and is more of a shell.  Though it is functional with a demo project.  I guess this makes sense for testing purposes...

con-f-use, 2017-07-28:
No idea what the nodemcu flasher is and spiffs, but whatever.

cnlohr, 2017-07-28:
We don't use spiffs. I think he's talking about the MPFS partition.   The NodeMCU Flasher is a GUI tool you can use to flash sections of memory on the ESP boards.

con-f-use, 2017-07-30:
Here are [exp82xx-project-v0.8-binaries.zip](https://github.com/cnlohr/esp82xx/files/1185963/exp82xx-project-v0.8-binaries.zip). I just build them and they are untested. I might make an 'official' release in the future, but don't count on it.

closed 2017-07-30 by con-f-use