cnlohr/esp82xx #71: Profiling of ESP8266 based code

yadavravi079, 2017-09-11:
Basic Infos
Hardware

Hardware: ESP8266
 Core Version:
 Description

Hi guys! I have developed a project based on NodeMCU, Now I am trying to debugging and profiling it.
 Can any could please help how to profile the code?
 I am able to debug it using gdb over eclipse and now I am trying to compile it using -pg Flag so that it gnenerate some info using mcount() and then I could use Gprof for further profiling.
 Share some ideas would be appreciated.
 Settings in IDE

Module: NodeMCU
 Flash Size: 4MB
 CPU Frequency: 80Mhz
 Flash Mode: dio
 Flash Frequency: 40Mhz
 Upload Using: SERIAL
 Reset Method: nodemcu

cnlohr, 2017-09-12:
I just noticed.  When you say based on NodeMCU, do you mean the hardware, or software as well?  If just the hardware and you're using C then it may work.

I don't think it would be very good, though.  It's not hard to set up an interrupt with a custom vector where you can get EXACTLY where the code was when it was interrupted and get a call listing of where time was spend BUT You would have to spend so long getting the data of where the code was. It would significantly slow down your process.  You'd also need a fair bit of RAM to store it, or stream it off very slowly over serial.

Is that acceptable?

yadavravi079, 2017-09-12:
Hardware is NodeMCU and software is esp8266 arduino core. I appreciate for your reply but and my idea is also quite similar which is that if I would be able to build my project using -pg Flag which direct the compiler to add some special calls to generated code for example: it will attach _mcount() around every instrumented function called and calculates the source and destination address and place it in the table. I will get the profiling information in the form of gmon.out which I call analyse using xtensa-lx106-elf-gprof.
I am not sure if the gmon.out will be generated on the microcontroller then do i need to do semihosting or it does have a file system so that i can get it directly instead i would need I/O library to get gmon.out file.

cnlohr, 2017-09-12:
Ooph.  I don't believe anyone's already developed a system to actually internally record or pull off the .out file.  I usually use LED profiling so I don't really know how I'd go about this other than the outline I mentioned above :(.  We can leave this here to see if someone else is interested.

gn0st1c, 2018-03-25:
@yadavravi079 : not the thing you're looking for, but [Remote Debug](https://github.com/JoaoLopesF/RemoteDebug) may help. 

`
/ /Debug.showProfiler(true); // To show profiler - time between messages of Debug
`