cnlohr/esp82xx #73: Failed to detect MPU9250 using I2C

csabakeszegh, 2018-02-03:
First, thank you for this great project. I'd like to connect to an MPU9250 using I2C to my NodeMCU clone. As the first step, I wrote a custom command `i` to enumerate I2C devices. I'm calling the following method:

```
#include "i2c_master.h"
int ICACHE_FLASH_ATTR scan_i2c(char *buf)
{
    int i = 0;
    int found = 0;
    i2c_master_gpio_init();
    for (i=1; i<127; i++) {
        i2c_master_start();
        i2c_master_writeByte(i<<1);
        if (i2c_master_checkAck()) {
            ++found;
        }   
        i2c_master_stop();
     }   
    ets_sprintf(buf, "Found i2c devices: %d", found);
}
```

The probIem that I always get 0 devices. The connection should be okay, because when I upload a sketch using Arduino IDE for Esp8266, utilizing the `Wire` library, it works and receive ACK for 0x68. I also tried a logic analyzer to track down the problem. 

When using `Wire` library it works:
[image]

When using `i2c_master` code I get NAK:
[image]

The difference I see that Wire is ~2x faster. (I tried to change the delays in i2c_master_wait, it became faster, but still got NAK.)

The MPU9250 device works well with an Arduino Nano (and also from the NodeMCU using the `Wire` library).

I tried to connect a Nano as I2C slave, and the enumeration worked with `i2c_master` code. Any help or hint is really appreciated.

cnlohr, 2018-02-04:
I... uhhh... I don't know what master_i2c.h is.  I've only used static_i2c.h. https://github.com/cnlohr/esplocalizer/blob/master/firmware-c/user/static_i2c.h

That said your trace looks good, so it's probably with your library.  You could try mine? 

Just 
```
#define I2CSPEEDBASE 1
#define I2CNEEDGETBYTE
#define DSDA 5
#define DSCL 4

#include "static_i2c.h"
```

Then in your code
```
ConfigI2C(); //Do this once...

	SendStart();
	r = SendByte( MPU_ADDY );
	if( r ) { SendStop(); printf("Fault\n"); return -3; }
	SendByte( ... ); //Do stuff
	SendByte( ... ); //Do other stuff. 
	SendStop();
```

csabakeszegh, 2018-02-07:
Thanks for the quick response. `i2c_master` is a software bit-banging I2C library in the [NonOS SDK](https://github.com/cnlohr/esp_nonos_sdk/blob/7d66ffc133f41cec6efd046ac04be70944a8b0ae/driver_lib/include/driver/i2c_master.h). By default it uses GPIO2 (SDA), and GPIO14 (SCL) pins. Changing them to GPIO4, GPIO5 respectively solved my problem. I'm not sure why. Maybe interfere with UART?

closed 2018-02-07 by csabakeszegh