cnlohr/esp82xx #76: mfsmaker / windows

king2, 2018-07-22:
I'm using unofficial SDK under windows, so compiling webpages is a pain (I need to reboot notebook to Linux, copy files, compile them, then copy them back, reboot back to Windows - just to understand that I have missed something important or made a mistake in javascript code).

Sad, but other tools have programs I'm using exists on Windows only, so I cannnot switch entire developing process to Linux :(

Is there any place where I can find mfsmaker.c compiled for windows?

Thanks!

e-neko, 2018-07-22:
Did you consider using Linux virtual machine, for example a virtualbox image? I've been using Ubuntu VMs for development for many years now, they can share folders/files with your Windows host, simplifying context switching immensely.

con-f-use, 2018-07-22:
The easiest way would be to develop in a Linux virtual machine, making use of shared folders between Windows host and Linux guest, if necessary. That way you could edit the html with Windows tools and get it to the esp with Linux.

The more complicated way is to cross-compile the binary. Usually when cross-compiling C code, I do it on Linux for a Windows target. For that I use mingw (`sudo apt-get install mingw-w64`). My guess is the line to invoke is something like:

```
x86_64-w64-mingw32-gcc -D_FILE_OFFSET_BITS=64 -D_DIRENT_HAVE_D_TYPE mfsmaker.c -o mfsmaker.exe
```

king2, 2018-07-22:
I will need to setup Linux VMs on all 3 notebooks just for compiling pages. I will better try to compile it :)

I did tried to use msys2 to compile it, but it tells me:
```
mfsmaker.c:61:12: error: 'struct dirent' has no member named 'd_type'; did you mean 'd_name'?
mfsmaker.c:61:21: error: 'DT_REG' undeclared (first use in this function)
```
even if I use -D_DIRENT_HAVE_D_TYPE (but I have used i686-w64-mingw32-gcc.exe).

e-neko, 2018-07-22:
You only need to prepare vm image once, then copy it over to all three notebooks. Alternatively, put it on a fast USB3 memory stick.

king2, 2018-07-22:
I have compiled mfsmaker.exe succesfully (removed DT_REG checking and added checking for "." and "..") - I have no folders inside webpage, so I can do it safely. 
It works and can compile webpage. Thank you for help!

closed 2018-07-22 by king2