cnlohr/esp82xx #78: esp_nonos_skd @ 80fbd83 missing

bbkiwi, 2018-08-04:
Hi Charles, I can't git submodule update since above is missing. 
Maybe it's just on you own local repo?
https://github.com/cnlohr/esp_nonos_sdk/tree/80fbd83de75ba6a8011704868b9a60ff888596f0

cnlohr, 2018-08-06:
It says it's a public repo.  I don't understand.  Can I see a more detailed error message?

bbkiwi, 2018-08-06:
Click 
https://github.com/cnlohr/esp_nonos_sdk/tree/80fbd83de75ba6a8011704868b9a60ff888596f0 and I get 404.

Look at both  branches your repo for esp_nonos_skd  by clicking 
https://github.com/cnlohr/esp_nonos_sdk/commits/dev and https://github.com/cnlohr/esp_nonos_sdk/commits/master. I don't see that sha there.

I also tried cloning and checking out dev and trying to update submodule and get
```
~/testclone` $ git clone --recursive https://github.com/cnlohr/esp82xx.git
Cloning into 'esp82xx'...
remote: Counting objects: 698, done.
remote: Total 698 (delta 0), reused 0 (delta 0), pack-reused 698
Receiving objects: 100% (698/698), 305.73 KiB | 243.00 KiB/s, done.
Resolving deltas: 100% (447/447), done.
Checking connectivity... done.
Submodule 'toolchain/esp_nonos_sdk' (https://github.com/cnlohr/esp_nonos_sdk.git) registered for path 'toolchain/esp_nonos_sdk'
Cloning into 'toolchain/esp_nonos_sdk'...
remote: Counting objects: 112, done.
remote: Total 112 (delta 0), reused 0 (delta 0), pack-reused 112
Receiving objects: 100% (112/112), 1.37 MiB | 595.00 KiB/s, done.
Resolving deltas: 100% (22/22), done.
Checking connectivity... done.
Submodule path 'toolchain/esp_nonos_sdk': checked out '7d66ffc133f41cec6efd046ac04be70944a8b0ae'
~/testclone $ cd e*
~/testclone/esp82xx (master) $ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
~/testclone/esp82xx (master) $ git branch -vv
* master b336d1c [origin/master] It is not worth the pain of showing passwords.
~/testclone/esp82xx (master) $ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/dev
  remotes/origin/ft-include_sdk
  remotes/origin/master
~/testclone/esp82xx (master) $ git checkout dev
M	toolchain/esp_nonos_sdk
Branch dev set up to track remote branch dev from origin.
Switched to a new branch 'dev'
~/testclone/esp82xx (dev) $ git diff
diff --git a/toolchain/esp_nonos_sdk b/toolchain/esp_nonos_sdk
index 80fbd83..7d66ffc 160000
--- a/toolchain/esp_nonos_sdk
+++ b/toolchain/esp_nonos_sdk
@@ -1 +1 @@
-Subproject commit 80fbd83de75ba6a8011704868b9a60ff888596f0
+Subproject commit 7d66ffc133f41cec6efd046ac04be70944a8b0ae
~/testclone/esp82xx (dev) $ git submodule update
fatal: reference is not a tree: 80fbd83de75ba6a8011704868b9a60ff888596f0
Unable to checkout '80fbd83de75ba6a8011704868b9a60ff888596f0' in submodule path 'toolchain/esp_nonos_sdk'
~/testclone/esp82xx (dev) $ git submodule sync
Synchronizing submodule url for 'toolchain/esp_nonos_sdk'
~/testclone/esp82xx (dev) $ git submodule update
fatal: reference is not a tree: 80fbd83de75ba6a8011704868b9a60ff888596f0
Unable to checkout '80fbd83de75ba6a8011704868b9a60ff888596f0' in submodule path 'toolchain/esp_nonos_sdk'
~/testclone/esp82xx (dev) $ 
```

con-f-use, 2018-08-06:
Seems like the `dev` branch of esp82xx (and only the `dev` branch) points to a commit of `esp_nonos_sdk` that has been deleted for that repository. Should be fixed now. No idea how that came to pass.

cnlohr, 2018-08-08:
I am so glad you know what you're doing with git, @con-f-use .  I get lost so quickly :(