cnlohr/esp82xx #86: Undefined reference to `os_printf`

Maistho, 2018-11-20:
Using `esp-open-sdk` with the patch for nonos-sdk v3.0 from this PR https://github.com/pfalcon/esp-open-sdk/pull/344

I first tried using the default version (2.1.0-18-g61248df) but that didn't work.

Running esp82xx from master, fresh install of everything

I'm assuming that I've missed something and forgot to include some `.o`-file somewhere, but after reading the docs I've come up with nothing. Did I simply miss some step?

Output from running `make burn`
```
$ make burn
/home/maistho/prog/esp8266/esp-open-sdk-v3/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DMFS_PAGE_OFFSET=65536      -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50 -DVERSSTR='"Version: 81ac6 - Build Tue, Nov 20 2018, 06:47:28 +0100 with -DICACHE_FLASH -DMFS_PAGE_OFFSET=65536      -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50"' -g esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib -L/home/maistho/prog/esp8266/esp-open-sdk-v3/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libmain.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/liblwip.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libssl.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libupgrade.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libnet80211.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libwpa.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libphy.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/liblwip.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libcrypto.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libc.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libespnow.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libpp.a /home/maistho/prog/esp8266/esp-open-sdk-v3/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/ld/eagle.app.v6.ld -T /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/ld/eagle.rom.addr.v6.ld -B/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib  -o image.elf
esp82xx/fwsrc/mfs.c: In function 'FindMPFS':
esp82xx/fwsrc/mfs.c:20:2: warning: passing argument 2 of 'ets_strncmp' from incompatible pointer type [enabled by default]
  if( ets_strncmp( "MPFSMPFS", mfs_check, 8 ) == 0 ) { mfs_at = MFS_PAGE_OFFSET; goto done; }
  ^
In file included from esp82xx/fwsrc/mfs.c:8:0:
/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/include/osapi.h:48:5: note: expected 'const char *' but argument is of type 'uint32 *'
 int ets_strncmp(const char *s1, const char *s2, unsigned int n);
     ^
esp82xx/fwsrc/mfs.c:25:2: warning: passing argument 2 of 'ets_strncmp' from incompatible pointer type [enabled by default]
  if( ets_strncmp( "MPFSMPFS", mfs_check, 8 ) == 0 ) { mfs_at = MFS_ALTERNATIVE_START; goto done; }
  ^
In file included from esp82xx/fwsrc/mfs.c:8:0:
/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/include/osapi.h:48:5: note: expected 'const char *' but argument is of type 'uint32 *'
 int ets_strncmp(const char *s1, const char *s2, unsigned int n);
     ^
user/user_main.c: In function 'user_init':
user/user_main.c:237:2: warning: passing argument 1 of 'ets_timer_disarm' discards 'volatile' qualifier from pointer target type [enabled by default]
  os_timer_disarm(&some_timer);
  ^
In file included from user/user_main.c:12:0:
/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/include/osapi.h:66:6: note: expected 'struct ETSTimer *' but argument is of type 'volatile struct ETSTimer *'
 void ets_timer_disarm(os_timer_t *ptimer);
      ^
user/user_main.c:238:2: warning: passing argument 1 of 'ets_timer_setfn' discards 'volatile' qualifier from pointer target type [enabled by default]
  os_timer_setfn(&some_timer, (os_timer_func_t *)timer100ms, NULL);
  ^
In file included from user/user_main.c:12:0:
/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/include/osapi.h:67:6: note: expected 'struct ETSTimer *' but argument is of type 'volatile struct ETSTimer *'
 void ets_timer_setfn(os_timer_t *ptimer, os_timer_func_t *pfunction, void *parg);
      ^
user/user_main.c:239:2: warning: passing argument 1 of 'ets_timer_arm_new' discards 'volatile' qualifier from pointer target type [enabled by default]
  os_timer_arm(&some_timer, 100, 1);
  ^
In file included from user/user_main.c:12:0:
/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/include/osapi.h:65:6: note: expected 'struct ETSTimer *' but argument is of type 'volatile struct ETSTimer *'
 void ets_timer_arm_new(os_timer_t *ptimer, uint32_t time, bool repeat_flag, bool ms_flag);
      ^
/tmp/cclb9OmE.ltrans1.ltrans.o:(.irom0.literal+0x128): undefined reference to `os_printf'
/tmp/cclb9OmE.ltrans1.ltrans.o: In function `cmd_GPIO':
/home/maistho/prog/esp8266/project/esp82xx/fwsrc/commonservices.c:312: undefined reference to `os_printf'
/home/maistho/prog/esp8266/project/esp82xx/fwsrc/commonservices.c:320: undefined reference to `os_printf'
collect2: error: ld returned 1 exit status
make: *** [esp82xx/main.mf:103: image.elf] Error 1
```

cnlohr, 2018-11-20:
You will need to `#include <osapi.h>` now.  This should be the case in the dev branch.  Can you see if this works with dev?  I need to do one more thing in dev before I can push it to master, but I'm almost there.

Maistho, 2018-11-20:
I received some other error when running the dev branch

If I replace `LoadDefaultPartitionMapprocTaskPrio` with `0` it builds and flashes! I'm assuming it should be some other value though.

```
make burn
/home/maistho/prog/esp8266/esp-open-sdk-v3/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DUSE_OPTIMIZE_PRINTF -DMFS_PAGE_OFFSET=532480 -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50 -DVERSSTR='"Version: b0c4a - Build Tue, Nov 20 2018, 17:46:58 +0100 with -DICACHE_FLASH -DUSE_OPTIMIZE_PRINTF -DMFS_PAGE_OFFSET=532480 -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50"' -g esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib -L/home/maistho/prog/esp8266/esp-open-sdk-v3/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libmain.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/liblwip.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libssl.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libupgrade.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libnet80211.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libwpa.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libphy.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/liblwip.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libcrypto.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libc.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libespnow.a /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib/libpp.a /home/maistho/prog/esp8266/esp-open-sdk-v3/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/ld/eagle.app.v6.ld -T /home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/ld/eagle.rom.addr.v6.ld -B/home/maistho/prog/esp8266/project/esp82xx/toolchain/esp_nonos_sdk/lib  -o image.elf
user/user_main.c: In function 'user_init':
user/user_main.c:219:27: error: 'LoadDefaultPartitionMapprocTaskPrio' undeclared (first use in this function)
  system_os_task(procTask, LoadDefaultPartitionMapprocTaskPrio, procTaskQueue, procTaskQueueLen);
                           ^
user/user_main.c:219:27: note: each undeclared identifier is reported only once for each function it appears in
make: *** [esp82xx/main.mf:129: image.elf] Error 1
```

cnlohr, 2018-11-20:
@AEFeinstein Do you recognize this?

cnlohr, 2018-11-20:
Whoops.  That's a typo.  Just fixed. 0734fb5438bc89de21dbd188195df83ef4e8cfe5

Maistho, 2018-11-20:
Nice, I've got it running now. Thanks for the help! :tada:

closed 2018-11-20 by Maistho

cnlohr, 2018-11-20:
I will merge dev into master as soon as I can!