cnlohr/esp82xx #93: [ webgui flashing ]

0xd3d0c3d, 2019-07-05:
new esp82xx flashing via webgui is nice, i had to make some changes a bit to work on 8285. 
i adjusted menuinterface.js to reflect 0x10000 rather than 0x40000 instances. i adjusted mpfs_start_at location because it *is* in different spot, you'll get file not found flashing page.mpfs otherwise. i also had to move the scratchpad location because if you flash bin pairs it will overwrite the page.mpfs and then you're left with no webgui unless you flash via esptool. this at least leaves the webgui ;p.. so firmware flashing is working gr8, even webgui flashing page.mpfs is great.. the issue is if you exceed 60k filesize. it will be missing data. every 65536 - 4096*2 will be 0xFF. :*[

// esp8285 - 1048576

[ user.cfg ]
    MFS_PAGE_OFFSET  = 532480
[ menuinterface.js ]
    // new esp82xx uses 0x00000 0x10000 fw bins, so adjusted all 0x40000 references to 0x10000
    var mpfs_start_at = 532480; // 65536;
    var flash_scratchpad_at = 655360; // 524288;
[ page.mpfs ]
    `make clean page.mpfs` // 512 328704
    `wc -c page.mpfs` // 329216 page.mpfs
[ flash ]
    esptool.py -b 115200 --port /dev/ttyUSB0 write_flash --erase-all -fs 1MB -fm dout 0xfc000 esp82xx/toolchain/esp_nonos_sdk/bin/esp_init_data_default_v08.bin 0x00000 image.elf-0x00000.bin 0x10000 image.elf-0x10000.bin 532480 web/page.mpfs
[ reflash ]
    // from webgui flash this same page.mpfs then dump it for a difference
[ dump ]
    // extracting reflashed page.mpfs from dump
    esptool.py read_flash 0x00000 0x100000 dump.bin
    dd if=dump.bin of=mpfs.bin bs=1 skip=532480 count=329216
[ compare page.mpfs versus mpfs.bin extraction ]
  - diff mpfs.bin page.mpfs
  // Binary files mpfs.bin and page.mpfs differ
  // ^- that's probably not good
[ visually compare the differences ]
  - vimdiff mpfs.bin page.mpfs
[ switch visual modes ]
  - :%!hexdump -C
  - ctrl + l
  - :%!hexdump -C
[ analyze differences ]
    >>> MPFS_FILE_SIZE=329216
    >>> offset1 = [ "0000e000", "57344" ]
    >>> hex(65536 * 1)
    '0x10000'
    >>> offset2 = [ "0001e000", "122880" ]
    >>> hex(65536 * 2)
    '0x20000'
    >>> offset3 = [ "0002e000", "188416" ]
    >>> hex(65536 * 3)
    '0x30000'
    >>> offset4 = [ "0003e000", "253952" ]
    >>> hex(65536 * 4)
    '0x40000'
    >>> offset5 = [ "0004e000", "319488" ]
    >>> hex(65536 * 5)
    '0x50000'
    >>> 57344 + 4096*2
    65536
    >>> 122880 + 4096*2
    131072
    >>> 188416 + 4096*2
    196608
    >>> 253952 + 4096*2
    262144
    >>> 319488 + 4096*2
    327680
    >>> hex(65536)
    '0x10000'
    >>> hex(131072)
    '0x20000'
    >>> hex(196608)
    '0x30000'
    >>> hex(262144)
    '0x40000'
    >>> hex(327680)
    '0x50000'

[image]
[image]

0xd3d0c3d, 2019-07-07:
i bypassed the issue with this code
```
function tohex8( c )
{
    var hex = c.toString(16);
    return hex.length == 1 ? "0" + hex : hex;
}

window.onload = function() {
    document.querySelector('input').addEventListener('change', function() {

    var reader2 = new FileReader();
    reader2.onload = function() {

        var arrayBuffer = this.result,
        array = new Uint8Array(arrayBuffer),
        binaryString = String.fromCharCode.apply(null, array);
        var mpfs = 532480;
        var sectors = Math.ceil(binaryString.length / 4096);
        var si = 0;
        var sector = mpfs / 4096;
        for (si = 0; si <= sectors; si++) {
            cmd = "FE" + sector;
            console.log(cmd);
            sector += 1;
            QueueOperation(cmd);
        }
        var rotations = binaryString.length / 256;
        var ri = 0;
        var bi = 0;
        var patch = "";
        for (ri = 0; ri < rotations; ri++) {
            patch="";
            console.log("rotation: " + (ri+1) + "/" + rotations)
            for (bi = 0; bi < 256; bi++) {
                bp = ri * 256;
                patch += tohex8(binaryString.charCodeAt(bp+bi)).toUpperCase();
            }
            cmd = "FX" + mpfs + "\t256\t" + patch;
            QueueOperation(cmd);
            mpfs += 256;
        }

    }
    reader2.readAsArrayBuffer(this.files[0]);

    }, false);
}
```

con-f-use, 2019-07-07:
care to make a pull request?

0xd3d0c3d, 2019-07-08:
give me a few days and i'll rewrite it into the format you guys already had written.

con-f-use, 2019-07-08:
That would be great. If it helps, there is a `CHIP` variable in `user.cfg` that gets passed down to other makefiles and is either `8266` or `8285`.

0xd3d0c3d, 2019-07-08:
[code](https://github.com/cnlohr/esp82xx/blob/master/web/page/menuinterface.js#L828)

the problem is the line 

```
if( ( pushop.place % flash_blocksize ) == 0 && flashresponse[1] != 'B' )
{
	QueueOperation( "FB" + ((pushop.place+pushop.base_address)/flash_blocksize),  function( x, y ) { ContinueSystemFlash( x, y, pushop ); } );
}
```
this works fine for image.elf-0x00000.bin, or image.elf-0x10000.bin because they align to 65536 evenly.<br>

##0x00000 = 0
##0x10000 = 65536

no problem.

##0x82000 = 532480
using 532480 does not divide evenly nor align properly

##>>> 532480 / 65536
##8.125

sending 'FB8.125' still works, but ideally it should be 'FB8', that .125 is equal to 8192 bytes.<br>
so the order operations is as follows.<br>

first pass.<br>

##pushop.place = 0
##pushop.base_address = 532480
##flash_blocksize = 65536

erase sector 8 or 8.125 in this case.<br>
##65536 * 8 = 524288
##524288 + 65536 = 589824
from 524288 to 589824 are ff'd, which is fine here..<br>
data from 0x0000 to 0xFFFF is written, 65536 blocks..<br>
when the pushop.place = 65536 the next sector is erased, herein lies the problem<br>
##532480 + 65536 =  598016
we wrote bytes from 532480 to 598016 and relied on a counter to 65536,<br>
because the alignment is wrong when we go to erase sector 9 we are clipping 8192 bytes.<br>
##65536 * 9 = 589824 
##598016 - 589824 = 8192
the data was actually written on the first pass, but it was erased on the second pass. ];><br>

i cheated and realigned things..<br>

#esp82xx/include/spi_memory_addrs.h
i changed 
```#define USER_SETTINGS_SIZE            0x3000```
to
```#define USER_SETTINGS_SIZE            0x1000```

this gives back 8192 bytes, and evenly aligns things.

#user.cfg
i changed
```MFS_PAGE_OFFSET  = 532480```
to
```MFS_PAGE_OFFSET  = 524288```

this compensates for the realignment in spi_memory_addrs.h

#web/page/menuinterface.js
i changed
```
var mpfs_start_at = 65536;
var flash_scratchpad_at = 524288;
```
to
```
var mpfs_start_at = 524288;
var flash_scratchpad_at = 655360;
```
also
```
if( ( pushop.place % flash_blocksize ) == 0 && flashresponse[1] != 'B' )
{
	QueueOperation( "FB" + ((pushop.place+pushop.base_address)/flash_blocksize),  function( x, y ) { ContinueSystemFlash( x, y, pushop ); } );
}

```
to
```
if( ( pushop.place % flash_blocksize ) == 0 && flashresponse[1] != 'B' )
{
	QueueOperation( "FB" + Math.floor((pushop.place+pushop.base_address)/flash_blocksize),  function( x, y ) { ContinueSystemFlash( x, y, pushop ); } );
}

```
also
- all instances of 0x40000 in menuinterface.js should be replaced with 0x10000, pointless otherwise.

***this keeps all the code written pretty much the same, it's updated to new esp82xx. you can flash <br>
firmware bins via wabgui now no problem, you can flash page.mpfs via webgui no problem. if you<br>
flash firmware bins via webgui it doesn't overwrite page.mpfs and vice versa. if i were to adjust<br>
the code, the only difference would be to make a boolean on the pushop to differentiate in the<br>
ContinueSystemFlash function to realize it's dealing with page.mpfs, and if so compensate for<br>
the 8192 byte offset differntial when dividing. anyways, its your code but i saw something i could<br>
fix.. thank you for you work.<br>***

0xd3d0c3d, 2019-07-08:
here's the pull request, again thank you for this code. :)
https://github.com/cnlohr/esp82xx/pull/94